kinectron/kinectron #11: Acquiring Data Real-Time

RinayShah, 2018-06-05:
On p5.js I was able to connect multiple kinetics and get the skeletons from both computers I was wondering is their is any way to acquire the data of the skeleton in real time from both kinetics? I want to fusion the skeletons as one and I need the dataset in order to do it.

lisajamhoury, 2018-06-05:
Kinectron does work in real time with two Kinects. Can you give me a little more detail about what your issue is?

RinayShah, 2018-06-05:
I want to acquire the data for the joints position of the skeleton in real time. Do know how i get that data or position of the joint of the skeleton in real time?

lisajamhoury, 2018-06-05:
[This example](https://github.com/kinectron/kinectron/blob/master/examples/p5examples/skeleton/sketch.js) accesses all of the joints in real time. See lines 39-60.

lisajamhoury, 2018-06-05:
Here's [an example in P5.js alpha editor](http://alpha.editor.p5js.org/lisajamhoury/sketches/H1t6wTggf). Just add your Kinectron IP address at line 22 and it should draw the joints when you run the sketch.

RinayShah, 2018-06-05:
I was able to get the skeltons from 2 Kinects onto P5.js however I am unable to get the exact coordinates of the skelton in real time. There must be a coordinate system on which these skeletons are mapped on I just want to be access those coordinates so as the hands or body changes directions the coordinates changes that’s the data I want to acquire.

lisajamhoury, 2018-06-05:
Each of the joints has several coordinates. It has color space x and y coords that correspond to the color image. Depth space x and y that correspond to the depth image. Then it has camera space points in x, y, z, which correspond to 3d space. Finally each joint has a 4-d joint orientation. Each of those coordinates is available as a property on the joint. For example, you can access joint.depthX and joint.depthY to get the x and y coordinates of the joint on the depth image. 

If you are using [this example code](http://alpha.editor.p5js.org/lisajamhoury/sketches/H1t6wTggf) log the joint at line 48 and you will see all of the coordinates on the joint.

RinayShah, 2018-06-06:
Thank you so much!

lisajamhoury, 2018-06-06:
No problem!

closed 2018-06-06 by lisajamhoury