kinectron/kinectron #12: Real-time obtain  x and y coordinates of the joint

jay-tailor, 2018-06-06:
I see that its possible to access joint.depthX and joint.depthY to get the x and y coordinates of the joint on the depth image, but how? Is there a way to see the coordinates in real time so as my hands moves the positions of the coordinates change?

lisajamhoury, 2018-06-06:
Hi @jay-tailor Take a look at the thread I just closed about this #11 

Does it answer your question?

lisajamhoury, 2018-06-06:
Try using [this example](http://alpha.editor.p5js.org/lisajamhoury/sketches/H1t6wTggf).

Add your Kinectron Server IP at line 22. And add 

```
console.log(joint);
debugger;

```
at line 48. 

When you run this, assuming you have data coming in from the Kinect, the code will stop and if you look in the console you will be able to see all of the properties on the joint. You can then access any of those properties with joint.[nameofproperty] so for example, joint.depthX or joint.cameraX.

jay-tailor, 2018-06-06:
This is what I got from the code alterations. Now wondering if there is a possible way to know which object represents which joint? Joints on Kinectron are ranged 0-24 (https://kinectron.github.io/docs/api2.html ), but the output on the console only gives them as objects. Also, what units are the positions in? 

[image]

RinayShah, 2018-06-06:
I wanted to know the same thing. What are the units are the position in? And how can we use the data to get coordinates of the joints since there are 25 joints in the skeleton. How can I use the objects data to get the individual data of the joints?

lisajamhoury, 2018-06-06:
There's a list of the joints and their corresponding numbers here: https://kinectron.github.io/docs/api2.html#accessing-individual-joints

When you use the Kinectron library, you can use the constants instead of the numbers, ie. kinectron.NECK instead of 2.

Here's a map of joints from MS SDK documentation:
https://i-msdn.sec.s-msft.com/dynimg/IC584844.png

About the coordinate system from the MS SDK Kinect documentation:

A color space point describes a 2D point on the color image. So a position in color space is a row/column location of a pixel on the image, where x=0, y=0 is the pixel at the top left of the color image, and x=1919, y=1079 (width-1, height-1) corresponds to the bottom right.
https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn758434(v%3dieb.10)

Depth space is the term used to describe a 2D location on the depth image. Think of this as a row/column location of a pixel where x is the column and y is the row. So x=0, y=0 corresponds to the top left corner of the image and x=511, y=423 (width-1, height-1) is the bottom right corner of the image. 
https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn758558(v%3dieb.10)

Camera space refers to the 3D coordinate system used by Kinect. The coordinate system is defined as follows:

The origin (x=0, y=0, z=0) is located at the center of the IR sensor on Kinect
X grows to the sensor’s left
Y grows up (note that this direction is based on the sensor’s tilt)
Z grows out in the direction the sensor is facing
1 unit = 1 meter

https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn772836(v%3dieb.10)

lisajamhoury, 2018-06-11:
Closing this now. Reopen if you have additional questions.

closed 2018-06-11 by lisajamhoury

jay-tailor, 2018-06-12:
We used console.log(joint); function and successfully got joint data from kinect # 1 streaming over a LAN to a different computer. The console shows real-time objects with color, depth, position as shown in the picture I sent few days ago. 

Now I'm having trouble getting data from 2 kinects streaming onto the same 1 computer over LAN. The problem is only one kinect is sending the joint data. We have to pause one kinect to send it from the other. Is there a way to have both kinect's sending the joint data (picture above) at the same time simultaneously? 

In other words, we are is there a way to get data from two exterior kinect-computer clients (at two different IP address, Slave) transferred to a 3rd computer (master) via p5.js.   

http://alpha.editor.p5js.org/jay.tailor/sketches/r16WbuplX
Have 2 kinects sending skeleton to a third computer. How can we get the JOINT data from those two kinects simultaneously one one console.

lisajamhoury, 2018-07-25:
I wrote this all out in a medium post. 

https://medium.com/@lisajamhoury/understanding-kinect-v2-joints-and-coordinate-system-4f4b90b9df16