kinectron/kinectron #14: multiple kinect joint data stream

jay-tailor, 2018-06-12:
We used console.log(joint); function and successfully got joint data from kinect # 1 streaming over a LAN to a different computer. The console shows real-time objects with color, depth, position as shown in the picture I sent few days ago.

Now I'm having trouble getting data from 2 kinects streaming onto the same 1 computer over LAN. The problem is only one kinect is sending the joint data. We have to pause one kinect to send it from the other. Is there a way to have both kinect's sending the joint data (picture above) at the same time simultaneously?

In other words, we are is there a way to get data from two exterior kinect-computer clients (at two different IP address, Slave) transferred to a 3rd computer (master) via p5.js.

http://alpha.editor.p5js.org/jay.tailor/sketches/r16WbuplX
Have 2 kinects sending skeleton to a third computer. How can we get the JOINT data from those two kinects simultaneously one one console.

jay-tailor, 2018-06-18:
We found a different way to approach this problem. Because the Kinect SDK uses C#, we require the Joint data to eventually be received by the C# program. The end goal is have the same skeleton displayed on Kinect's SDK with multiple kinects for multiple angles. 

we are looking into the Kinectron.exe application and it's code. We recognized a couple of functions that are of interest in respects to sending JOINT data over a network. Would you know a possible method to send data over to be received by a c# code using Kinectron.exe. Thanks for all your help aswell. Pictures attached are code we think can be slightly altered to do just that. 

[image]
[image]

lisajamhoury, 2018-06-18:
Kinectron is built on top of the [Kinect2 node library](https://github.com/wouterverweirder/kinect2). @wouterverweirder did the heavy lifting to bring the Kinect data from the SDK into a Javascript environment. You will have to make alterations there in order to stream c#.  If you want to work in c# though, I'd recommend working in Visual Studio, and using a UDP / socket server and client directly from there. It will be much faster to go direct since you don't need JS.

jay-tailor, 2018-06-20:
Do you know what kind of alterations would be ideal to get it streaming to C# as you say? Also, would a C# code under a UDP/Socket connection be able to obtain joint data from another computer on the same LAN (not needing p5.JS) ?

lisajamhoury, 2018-07-02:
No I don't really know, but Kinectron is not the right tool for that. I don't know of another tool that does that, but Kinectron is made specifically for Javascript and to be used by client in the browser. Sorry I can't help more.

closed 2018-07-02 by lisajamhoury