kinectron/kinectron #15: Error resolving paths in Macbook's Bootcamp

Jagni, 2018-06-19:
Uncaught Error: The specified module could not be found.
\\?\C:\kinectron-server-0.0.5.1\resources\app\node_modules\kinect2\build\Release\kinect2.node

I'm using the preview release 0.0.5.1 and got this error after starting kinectron-server.exe

I've got a Windows 10 dual boot in a MacBook Pro Mid 2014 using Bootcamp.

lisajamhoury, 2018-06-19:
See #7 

Does that help?

Jagni, 2018-06-19:
Unfortunately no...

I've deleted the directory, re-downloaded [the file](https://github.com/kinectron/kinectron/releases/download/0.0.5.1/kinectron-server-0.0.5.1.zip) and unzipped it in C:\

But I'm still going through the same error

Jagni, 2018-06-19:
Okay, it turns out that I only had kinect SDK 1.0 and 1.8 installed. Installing 2.0 solved it!
Sorry for the mistake

lisajamhoury, 2018-06-20:
Great!

closed 2018-06-20 by lisajamhoury

pesinasiller, 2020-08-17:
any updates on this? i have a similar problem
I have a Windows 10 dual boot in a MacBook Pro Mid 2014 using Bootcamp and kinect SDK 2.0

Uncaught Error: The specified module could not be found.
\\?\C:\kinectron-server-win32-x64\resources\app\node_modules\kinect-azure\build\Release\kinectAzure.node

lisajamhoury, 2020-08-17:
Which release of the server are you using? Where are you seeing this error -- Is it in electron console on open? Are you planning to use the Kinect v2 or the Azure Kinect? 

I haven't personally tested on bootcamp, but I know people have used Kinectron successfully on bootcamp, so hopefully we can figure it out.

pesinasiller, 2020-08-19:
I'm trying to use release 0.3.3 and kinect 2. Yes, the error is showing in the electron console on open. I searched the file mentioned and it is there, in that path.