kinectron/kinectron #28: Versioning system for client-side api

lisajamhoury, 2018-07-12:
What is the proper way to version the client library? I add version numbers when I create new releases of the server, but there's currently no way to know which client library file is the most up to date.

montoyamoraga, 2018-07-13:
I say we make them match, what do you think?

lisajamhoury, 2018-07-18:
@montoyamoraga It wouldn't work to match them since I develop them at different times

montoyamoraga, 2018-07-18:
oh i see, so then different numbers should be fine

lisajamhoury, 2018-07-18:
ok. do you know of any good examples. ie how this should be noted in the
file? i can look it up, just thought you might already have good examples.

montoyamoraga, 2018-07-19:
I want kinectron-clien.js to console.log on the terminal when it's loaded,
including the version number, just like Tone.js does it.
You can see an example here, when you open the console
[Tone.js example](https://tonejs.github.io/examples/#envelope)

Also, the file should be minified for using Pierre's solution for CDN

lisajamhoury, 2019-11-10:
Create a success message on load as part of the logged message as well.

lisajamhoury, 2019-12-01:
This is included in version 0.3.0

closed 2019-12-01 by lisajamhoury