kinectron/kinectron #30: How to use Peer ID to get Kinect Joint Data

RinayShah, 2018-07-16:
I want to use the Peer ID to get the kinect joint data is that possible and if so how can it be done?
[image]
[image]

lisajamhoury, 2018-07-16:
The joint data is passed in as an argument in the callback that you set when you start tracked bodies. So in your case, the joint data is a property on "body" at line 56, the funtion bodyTracked. The 26 joints of the Kinect skeleton are in body.joints. See #12 for more info on joints.

lisajamhoury, 2018-07-18:
Closing this, but you can reopen if you have a more specific question.

closed 2018-07-18 by lisajamhoury