kinectron/kinectron #35: console.log multiple arguments

jay-tailor, 2018-07-23:
Modified Code --> http://alpha.editor.p5js.org/jay.tailor/sketches/SkZ2_s7V7

Since we can see two bodies (1 from each kinect) on one screen, this means the joint data is being sent over. We wish to see both kinect's joint data on console.log. This way we can extend the program to more than 2 kinects later easily. Do you know how where we can put >> console.log(joint, joint2); << so it can display both joints data? 

2 different kinect, 1 screen, over same network --> 
[image]

lisajamhoury, 2018-07-24:
You will have to use global variables so that you can log them at the same time. Here are the lines of code to add into the functions that you have so far. 

```
// add global variables 
var joints1 = null;
var joints2 = null;

function bodyTracked(body) {
  // assign value to joints1
  joints1 = body.joints;
}

function bodyTracked2(body) {
  // assign value to joints2
  joints2 = body.joints;
}

function draw() {
  // log current values at the same time
  console.log(joints1, joints2);
}

```

closed 2018-07-24 by lisajamhoury

lisajamhoury, 2018-07-24:
console.log('joints1: ' +  joints1);
console.log('joints2: ' + joints2);