kinectron/kinectron #39: server

profjere, 2018-10-25:
I've just started using the Kinectron.  I can get everything to work fine when I'm using the p5.js editor on the same computer as the server (and have this line):
kinectron = new Kinectron();

When I add the IP address on another computer on our home network, with otherwise the same code, it doesn't get the feed. (I switched to a mac for this.)  So, first question: what can cause that?

And second question, how do you set up the peer network on a mac.  That seems like the next step?

lisajamhoury, 2018-10-30:
Are you getting any errors in the JS console of the browser on your Mac? A screenshot of the console on the browser would help assess what's going on. 

What browser are you using? This is really only tested in Chrome.

Did you try the p5.js alpha editor on the Mac with the IP address? Any errors there? 

The peer network is automatically setup on your local network. Setting up your server to run on a public network is a bit more involved. Here's more information on the options with the peer server https://kinectron.github.io/docs/server3.html#peer-server

profjere, 2018-11-04:
It all works fine on one computer, including working in the p5.js editor without calling the IP address.

However, when I add the IP address on either that same PC or my mac (sitting next to the computer), I get the same error:

kinectron.bundle.js:1084 Mixed Content: The page at 'https://editor.p5js.org/jsarachan/sketches/BJRVOuJhm' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://192.168.0.24:9001/peerjs/id?ts=15413438537140.5700176313907461'. This request has been blocked; the content must be served over HTTPS.

So, it's related to https and security?

lisajamhoury, 2018-11-04:
What is the link you are using to include the Kinectron library in your project? ie. The script line from index.html file.

What is the exact line of code you are using at kinectron.makeConnection()?

You can just sent me a link to your public p5 code in the editor if you'd like.

profjere, 2018-11-04:
This is very kind of you.  Here's the link:

https://editor.p5js.org/jsarachan/sketches/BJRVOuJhm

p5.js Web Editor<https://editor.p5js.org/jsarachan/sketches/BJRVOuJhm>
editor.p5js.org
A web editor for p5.js, a JavaScript library with the goal of making coding accessible to artists, designers, educators, and beginners.

Jeremy Sarachan
Chair and Associate Professor, Department of Media and Communication
Director, Interactive Media

Office: Fay 210
--
3690 East Ave.
St. John Fisher College
Rochester, NY 14618
585-385-7277

lisajamhoury, 2018-11-05:
No problem. I didn't get a chance to test today with a live Kinectron server, but I will try to get to it late this week. It is possible it is a change to the p5 editor that is causing the error with the server from local host.

The way to know if it's the p5 editor that is causing the problem is to run the p5 sketch from your desktop with a local server. Here are [instructions](https://github.com/processing/p5.js/wiki/Local-server) for that from the p5 wiki. 

Here's something else to try. It's a long shot, but it might be worth trying since we were having some CORS errors a while back with the ITP server: 

Try putting this

`<script scr="https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.0.5.1/client/bin/kinectron.bundle.js"></script>`

Instead of this: 
`<script src="https://itp.nyu.edu/kinectron/kinectron.bundle.js"></script>`

on line 7 of index.html. 

LMK how it goes.

montoyamoraga, 2018-11-05:
> Are you getting any errors in the JS console of the browser on your Mac? A screenshot of the console on the browser would help assess what's going on.
> 
> What browser are you using? This is really only tested in Chrome.
> 
> Did you try the p5.js alpha editor on the Mac with the IP address? Any errors there?
> 
> The peer network is automatically setup on your local network. Setting up your server to run on a public network is a bit more involved. Here's more information on the options with the peer server https://kinectron.github.io/docs/server3.html#peer-server
> 
> Also, @montoyamoraga was working on writing instructions for setting up your own peer server, but I don't know how far along he is. Maybe he can update here.

Hi @lisajamhoury 
Sorry about the delay, I have made some progress about  peer servers, but got stuck in how to create your own certificates for https and stuff, i will upload what i have so far on a new repo =)

and @profjere let us know if you had success following lisa's instructions, happy to help

profjere, 2018-11-06:
Thank you--the new script link didn't work--in fact, it gave me an error with the first Kinectron line, so I don't think it was connecting to the library at all.

I will check out the local server idea in the next couple of days.

Jeremy

Jeremy Sarachan
Chair and Associate Professor, Department of Media and Communication
Director, Interactive Media

Office: Fay 210
--
3690 East Ave.
St. John Fisher College
Rochester, NY 14618
585-385-7277

profjere, 2018-11-06:
Thank you--the new script link didn't work--in fact, it gave me an error with the first Kinectron line, so I don't think it was connecting to the library at all.

I will check out the local server idea in the next couple of days.

Jeremy

nicolaspe, 2018-11-10:
Hi all!

I encountered the same error today and it totally seems to be something about security. The p5 editor is served over HTTPS, but the Kinectron server is not. I used python's http.server and runs perfectly that way, same code and everything (using ITP's link). 

And same as @profjere , using the cdn link for kinectron (not itp's) breaks everything.

lisajamhoury, 2018-11-27:
@pdr will you look at cdn link?

I will take a look at p5 editor integration. I've run into CORS errors there in the past with other libraries. May be related.

lisajamhoury, 2020-08-23:
I just pushed release 0.3.4 with a fix for this. I've added an option to access the Kinectron server with ngrok. It works by creating an https tunnel to the Kinectron server at your localhost port 9001 (127.0.0.1:9001).

Ngrok uses a public url. This means that in addition to https, which is what the p5 Web Editor requires, you now have a Kinectron server that can be accessed from anyone on the internet. (@profjere I think this provides a new solution to your original question.)

Opening the Kinectron server to the internet does mean that anyone with the url could possibly exploit security vulnerabilities in the Kinectron server and its dependencies. (I'm not saying there are any vulnerabilities, but there may be...) If you're just using this for local development the risk of exploitation should be very low because ngrok creates a unique alphanumeric url that changes each time the server is run. (This is good for safety, although will be a bit annoying as the developer bc you need to replace the public address in the client every time you restart the server. ) So there is little likelihood that someone with malicious intent would be able to find you. If you'll be publicly sharing your public Kinectron url, you'd be opening yourself up to higher risk (although still relatively low).

Instructions are here: "Create an Instance of Kinectron." But essentially all you do is click the button for "Create Public Address" and then use that address in the client when creating an instance of Kinectron instead of the ip address.

let kinectron = new Kinectron("46b028e5c8ec.ngrok.io");

I'm going to close this issue for now, but please feel free to comment and let me know how if it is working for you, if you have any questions, etc. Thanks!

closed 2020-08-23 by lisajamhoury