kinectron/kinectron #42: Can't access the kinect data from from another computer

DannyRoberts95, 2019-01-16:
I'm having a problem with getting the kinect data to pass from my PC over to my Mac. The kinectron data is picked up in browser no problem when I try on the PC that the kinect is connected to. 
But when I try to access the kinectron server from my MAC by passing in the IP address there is no response. Could this be to do the college firewall perhaps?

Here is my code.
---------------------------------------------------
let kinectron;

function setup() {
  createCanvas(400, 400);
	kinectron = new Kinectron("172.16.31.39");
	kinectron.makeConnection();
	kinectron.startTrackedBodies(drawBodies);
}

function draw() {
  background(220);
}

function drawBodies(body){
console.log(body)
}
------------------------------------------------------

Thanks!!

lisajamhoury, 2019-01-16:
A few thoughts:

- Try deleting your draw function entirely
- Double check the PC and Mac on the same network? 
- What (if any) errors are you getting in your JS console in the browser? (Send a screenshot if possible)

DannyRoberts95, 2019-02-12:
Sorry for the delayed response. I'll try out those fixes and if I still have any trouble I'll post them here. Thanks a mill for your speedy response!

lisajamhoury, 2019-02-25:
I'm closing this now, but please reopen if you have more details on the problem.

closed 2019-02-25 by lisajamhoury