kinectron/kinectron #43: connection not working

muhammadUmairKhankaker, 2019-01-30:
I am doing everything according to tutorial, but my code doesn't show any output. I am running this example [https://www.youtube.com/watch?v=BV6xK3EOznI](url) code on the same machine as the server. 
there is no output in the console or canvas.

lisajamhoury, 2019-02-04:
Do you have output on the canvas or in the console of the server application?

muhammadUmairKhankaker, 2019-02-04:
Thanks for the reply Lisa. There was no output on either canvas or console.
Its resolved now.
I was running the code on firefox before. Just switched to chrome and it
worked perfectly. Not sure why it wasnt working on firefox.
But i have noticed that the feed on the canvas has a bit of lag even though
the server feed is smoooth.

lisajamhoury, 2019-02-13:
Ok glad you got it to work. The lag may be a result of the peer server library that I am using in Kinectron. It is a few years out of date and needs to be replaced.

closed 2019-02-13 by lisajamhoury