kinectron/kinectron #45: Getting feed on server but not sending to P5 local editor

horizonqu, 2019-03-29:
Kinectron is getting feed(showing skeletons) on server, but example code are not receiving data from kincectron.

The two laptops are on the save WiFi. One thing that is also strange: the IP address in Kinectron is not updating, always showing the same address when I restart the software.

The whole setup worked yesterday under the same environment and network.

Any idea how could I fix it?
Thanks!

rayuli, 2019-09-09:
Hi, I have exactly the same problem. 
thanks!

lisajamhoury, 2019-09-11:
Can you provide more information about the problem? Are you getting any errors in the console on either the server or the client? Sending a screenshot of the server interface and open console could also help.

rayuli, 2019-09-11:
Hi! 

The kinectron server is geeting feed and doesn´t show any error messages. The p5.js online editor doesn't get any data and doesn't show any message. It doesn't print either the console.log() message.
I have tried this same example using a local web server for chrome and works perfect. 

thanks for answering!

J

[image]

rayuli, 2019-09-14:
Hi!
Yes, it shows several errors.
I´ll keep using a local server. 
thanks for answering!
[image]

tbenbow, 2019-09-25:
I have an app that was running for several months and lost the ability to connect after a Chrome update. I was getting warning similar to the ones above about RTCIceServer.url and onaddstream  being deprecated. I downgraded to Chrome v64 and my app gained connectivity again.

lisajamhoury, 2019-09-25:
Ok good to know. I'll take a look at this before the end of the week.

lisajamhoury, 2019-09-27:
It looks like the main issue here is the [mixed content error](https://developers.google.com/web/fundamentals/security/prevent-mixed-content/what-is-mixed-content), which I'm getting in both Firefox and Chrome. I will dig into how to resolve this, but I surfaced a lot of other issues while digging into this. 

In Chrome, Peerjs is not firing connection.open on the application side when it receives a peer connection. This is preventing the peer connection from ever sending data between app and client. This is [a known issue ](https://github.com/peers/peerjs/issues/293), but it may have been addressed in recent Peerjs updates.  

I need to double check that PeerJS is being kept up to date in dependencies. But long term we need to move off of Peerjs entirely because it's not being maintained consistently. I will touch base with @js6450 about options and open a new issue for this. 

Also, the NYU kinectron bundle link is getting a CORS error again. Need to fix with @vanevery. Long term, I will update all example links to jsdelivr and check that development pipeline is changed to keep jsdelivr link up to date. 

For RTCIceServer warning, need to look back at #9

lisajamhoury, 2019-11-10:
I just published a new release of Kinectron server and application, which solves many of the connectivity issues between the server and the client. 

The new server is available at https://github.com/kinectron/kinectron/releases/tag/0.2.0 and the client is at:

https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.2.0/client/dist/kinectron-client.js

Note that the changes will only work with a local server (like SimpleHTTPServer), not from the p5 online editor, which I hope to have running again in the next few weeks.

slowizzm, 2019-12-08:
Curious if there was an update with this working in the web editor? No rush just noticed it is still getting the Mixed Content err.

lisajamhoury, 2019-12-09:
Thanks for checking in. Not yet, but I hope to tackle this or next week. I'll update here when there's news.

slowizzm, 2020-03-30:
Any updates on this? I noticed this is still not working, thought I would check in ...

lisajamhoury, 2020-04-03:
Thanks for checking in. This is still not working. This is an issue with Chrome and Firefox blocking insecure connections to the local network. I have researched a number of solutions for this, but haven't yet found anything that will work as Chrome and Firefox are intentionally blocking this type of connection. 

If you happen to be working with both the p5 web editor and the Kinectron server on the same computer, you can use 127.0.0.1 as the IP address in the web editor and it will work. 

`let kinectronIpAddress = "127.0.0.1";`

lisajamhoury, 2020-08-23:
I just pushed release [0.3.4](https://github.com/kinectron/kinectron/releases/tag/0.3.4) with a fix for this. I've added an option to access the Kinectron server with ngrok. It works by creating an https tunnel to the Kinectron server at your localhost port 9001 (127.0.0.1:9001).  

Ngrok uses a public url. This means that in addition to https, which is what the p5 Web Editor requires, you now have a Kinectron server that can be accessed from anyone on the internet. 

Opening the Kinectron server to the internet does mean that anyone with the url could possibly exploit security vulnerabilities in the Kinectron server and its dependencies. If you're just using this for local development the risk of exploitation should be very low because ngrok creates a unique alphanumeric url that changes each time the server is run. (This is good for safety, although will be a bit annoying as the developer bc you need to replace the public address in the client every time you restart the server. ) So there is little likelihood that someone with malicious intent would be able to find you. If you'll be publicly sharing your public Kinectron url, you'd be opening yourself up to higher risk (although still relatively low). 

Instructions are here: "[Create an Instance of Kinectron](https://kinectron.github.io/docs/api.html#2-create-an-instance-of-kinectron)." But essentially all you do is click the button for "Create Public Address" and then use that address in the client when creating an instance of Kinectron instead of the ip address. 

`let kinectron = new Kinectron("46b028e5c8ec.ngrok.io"); `

I'm going to close this issue for now, but please feel free to comment and let me know how if it is working for you, etc. Thanks!

closed 2020-08-23 by lisajamhoury