kinectron/kinectron #47: kinectron.startTrackedBodies() not working

Diji19, 2019-10-03:
kinectron.makeConnection() is working and I get a peer ID , but when I start the function kinectron.startTrackedBodies(drawSkeleton) the callback won't start.

lisajamhoury, 2019-10-03:
Yes this is a known issue in Chrome with Peerjs. Peerjs is not firing connection.open on the application side when it receives a peer connection. This is preventing the peer connection from ever sending data between app and client. I am working on resolving this, but in the meantime it seems to not be an issue in Firefox. Please use Firefox for your client side for now.

Diji19, 2019-10-03:
I tried it know in Firefox, but it seems I have the same probleme in there!

lisajamhoury, 2019-10-03:
Please post any errors you are getting in the browser JS console.

Diji19, 2019-10-03:
RTCIceServer.url is deprecated! Use urls instead. kinectron.js:1736
RTCIceServer.url is deprecated! Use urls instead. kinectron.js:1761
RTCIceServer.url is deprecated! Use urls instead. kinectron.js:1779
Der Orientierungssensor sollte nicht mehr verwendet werden. p5.js:55419:23
Der Bewegungssensor sollte nicht mehr verwendet werden. p5.js:55419:23
RTCIceServer.url is deprecated! Use urls instead. kinectron.js:722
onaddstream is deprecated! Use peerConnection.ontrack instead. kinectron.js:797
My peer ID is: 5a3au39y9z6t7o86 kinectron.js:3061:13

My code is from Getting Started: API

lisajamhoury, 2019-10-03:
Many of these are warning and shouldn't be keeping the code from running. Can you send a screenshot so I can see what is actually an error (red) versus a warning (yellow). Some of these are in German I think so I'm not sure what they are. 

Are you working in the p5 online editor or from your local host?

Diji19, 2019-10-04:
[image]

I am working from my localhost and I have no errors!

lisajamhoury, 2019-10-04:
Can you post your code? Are you getting errors on the server application (the electron application)?

Diji19, 2019-10-10:
`// Create a p5 canvas (learn more at p5js.org)
var myCanvas = null;

// Declare kinectron
var kinectron = null;

function setup() {
  // Create a p5 canvas
  myCanvas = createCanvas(500, 500);

  // Set background color
  background(0);

  // Initialize Kinectron
  initKinectron();
}

function initKinectron() {

  // Define and create an instance of kinectron
  kinectron = new Kinectron("10.18.250.25");

  // Connect with server over peer
  kinectron.makeConnection();

  // Request all tracked bodies and pass data to your callback
  kinectron.startTrackedBodies(drawSkeleton);
}

// The incoming "body" argument holds the Kinect skeleton data
function drawSkeleton(body) {

  // Clear the background
  background(0, 20);

  // Draw a circle at the location of each joint
  for(var i = 0; i < body.joints.length; i++) {

    // Get the joint
    var joint = body.joints[i];

    // Set the drawing color
    fill(100);

    // Map Kinect joint data to canvas size; Draw the circle
    ellipse(joint.depthX * myCanvas.width, joint.depthY * myCanvas.height, 15, 15);
  }
}
`

[image]

lisajamhoury, 2019-10-14:
Thanks for posting your code. This appears to be an issue with the peerjs library that is responsible for sending data between the server and the browser. I am working to find a solution and will post here when it's resolved.

lisajamhoury, 2019-11-10:
Hi @Diji19 

I just published a new release of Kinectron server and application. I believe this should fix the problems that you are having.

The new server is available at https://github.com/kinectron/kinectron/releases/tag/0.2.0 and the client is at:

https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.2.0/client/dist/kinectron-client.js

Note that the changes will only work with a local server (like SimpleHTTPServer), not from the p5 online editor, which I hope to have running again in the next few weeks.

Please re-open the issue if it doesn't work for you.

closed 2019-11-10 by lisajamhoury