kinectron/kinectron #49: Kinectron.server doesn't work

Hongiee2, 2019-10-30:
Thanks to Kinectron, I made a great piece of work last semester.

Few Months later, I tried to operate same task but it doen't work.

connecting with kinectron in p5.js is working well but data(joint, image) doesn't come in my p5.js

I'd appreciate your help. 

Sorry for not being able to speak English well.

lisajamhoury, 2019-10-31:
Hi! Thanks for using Kinectron! Were your able to get your project running again? 

There is currently an issue with the underlying connection between p5 in the browser and Electron. I think I found a solution, but just need to implement and test it — hopefully it will be working in a week or two.

lisajamhoury, 2019-11-10:
Hi @Hongiee2 

I just published a new release of Kinectron server and application. I believe this should fix the problems that you are having. 

The new release is available at https://github.com/kinectron/kinectron/releases/tag/0.2.0 and the client is at: 

https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.2.0/client/dist/kinectron-client.js

Note that the changes will only work with a local server (like SimpleHTTPServer), not from the p5 online editor, which I hope to have running again in the next few weeks.

Hopefully this will solve your issues. Please re-open the issue if it doesn't work for you.

closed 2019-11-10 by lisajamhoury

Hongiee2, 2019-11-10:
Thank U for your effort

I'll confirm soon

 I really appreciate it !

@lisajamhoury