kinectron/kinectron #56: smaller size of kinectron-server to avoid dropbox

montoyamoraga, 2019-12-03:
@lisajamhoury i just built the app on my computer and the folder kinectron-server-win32-x64 is 1.8 GB uncompressed, and 1.3 GB compressed, so it should be possible to add to the release, and avoid dropbox, right?

maybe i am missing something.

lisajamhoury, 2019-12-06:
Look at #3 in Contribute.md under [contribute to the server application](https://github.com/kinectron/kinectron/blob/master/CONTRIBUTE.md#contribute-to-server-application).

It appears to be a requirement of the Azure Body Tracking SDK that these files are in the root of the project to function. They are not committed to github because they are too big in size and they are generated in the azure kinect node module on npm install. With these files the build is too big to upload  to github. I considered separating the models out to a separate download and instructing users to download and include them in the unzipped Kinectron folder. But then users would still be needing to download something from dropbox (or elsewhere), so I think it's easier for users to just download the fully bundled ready to use file in one place.

lisajamhoury, 2019-12-28:
Closing this for now, but please re-open if there is action we can take here.

closed 2019-12-28 by lisajamhoury