kinectron/kinectron #61: Create new example — Skeleton With Images

lisajamhoury, 2020-01-14:
Create a new example using Kinect joints to move images representing bones in 2D. Create first for Windows, then for Azure.

lisajamhoury, 2020-01-14:
Just pushed [a new branch](https://github.com/kinectron/kinectron/tree/skeletonimages) with this new example. 

@montoyamoraga Do you have time to check the example and refactor to make more useable/understandable? You should be able to work directly on the kinectron/skeletonimages branch. 

The example currently uses recorded data. Do you think it's a good idea to leave it running from Kinectron recorded data? I do this for the documentation website examples, but not sure if it's a good idea for the in-repo examples. 

Also the recorded data isn't great on the legs. Should I record new data, or do you think this is ok as is?

montoyamoraga, 2020-01-15:
ok perfect, will do it today after lunch, will post here when i have some comments and tag you :)

montoyamoraga, 2020-01-15:
hi!

(this part is for me) i did the following steps to pull your new branch:
* on my fork, created a branch with the same name as your new branch: skeletonimages.
* did a PR on my fork to merge the commits on your branch on mine.

this part is for @lisajamhoury, i did the following improvements on the example:
1. on the .html file, i upgraded to latest p5.js with their CDN.
2. on the .html file, i discovered a way to use the same CDN that p5.js uses, that allows to distribute files from GitHub releases, so instead of having the old style
"./../../../../client/dist/kinectron-client.js" to load the file, i switched it to
"https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.3.1/client/dist/kinectron-client.js".
here is more info: https://www.jsdelivr.com/?docs=gh
3. on the js file, i added mostly comments, like the text i wrote for ip address that i included on the gettingstarted example, and i also added a header for mit license and kinectron copyright.
4. on the js file, i added brackets when you have if statements with only one line on the body, any strong preference to write without brackets? i rather be verbose :)
5. on the js file, to address the live data vs pre-recorded data, on setup(), i added some logic to print on console info about which mode it is running now, and how to switch to the other. i like it that is using pre-recorded data at first, i rather have that as the default, yay.

i tested everything and it ran great. i would leave it as it is now with my comments, and think about it. i think our examples should be short and straight to the point, and this one is great :)

i loved the comments with links to push() and pop(), i was going to add a comment to it, but you had the link yay, and there are also links to your medium articles, perfect!

lisajamhoury, 2020-01-16:
Excellent. Thank you. Just merged all your changes with the updates you made to 2 and 5. 

I'll now work on building this example out for Azure. Will merge with master and add to documentation website once it's complete.

lisajamhoury, 2020-01-17:
@montoyamoraga I added the azure version of example. I know you don't have azure device, but if you have a moment it'd be great if you can take a look at the example using recorded data. I had to add some things to the sketch file for normalizing joints. Would appreciate if you could take a look at it and let me know if it's ok, or needs any improvements. Thanks!

montoyamoraga, 2020-01-18:
@lisajamhoury yeah i don't have an azure, but i went over the changed files on the branch skeletonimages and they look great to me.
all the things we discussed, such as the brackets for if statements with one line in their bodies, are there.
i would merge it, and i think something more long term i should do this semester, is:
* go over all the examples, test them with the newest p5.js
* write a style guide for kinectron, with helpful reminders about how to write code in both the src and the examples.
* a template for the examples, which can be copy/pasted and then repurposed for further examples.

lisajamhoury, 2020-01-20:
Great thanks for looking! Just released a new release of the server and api that includes record support for Azure and this example. I also added a new record example for Azure. I'll open a new issue for that. 

Your to-do list looks great. A template would be really helpful.

closed 2020-01-20 by lisajamhoury