kinectron/kinectron #63: p5 Error on text()

lisajamhoury, 2020-01-17:
I'm encountering an error I haven't seen before when I try to use the p5 text() functionality. 

@montoyamoraga Have you seen this before? Any idea what might be causing it? I just wanted to use the functionality for debugging, so it's not imperative to figure out. But I'm curious.

When I add text() to [sketch.js in the new example](https://github.com/kinectron/kinectron/blob/skeletonimages/examples/windows_examples/p5_examples/skeletonimages/sketch.js) at line 331, like this

```
// draw skeleton
function drawKinectJoint(joint) {
  // drawing settings
  noStroke();
  fill(100);

  push();
  translate(width / 2, height / 2);
  ellipse(joint.cameraX * skelWidth, joint.cameraY * skelHeight * -1, 15, 15);
  text("hello", 10, 30);
  pop();
}
```

I get the following error in my console: 

“Uncaught SyntaxError: Cannot use import statement outside a module” 

The error happens also in my azure code. 

Thanks!

https://p5js.org/reference/#/p5/text

montoyamoraga, 2020-01-17:
@lisajamhoury 
does it also happen with the previous p5.js version?

also, i just did a quick test, using text() on a 2D sketch, and it worked fine.
but then i tried with WEBGL, a 3d sketch, and it broke because it wanted me 
to have loaded a font first.

montoyamoraga, 2020-01-17:
the examples on the p5.js editor use the latest p5.js version, 0.10.2,
here is the example i broke with the 3d sketch.
https://editor.p5js.org/montoyamoraga/sketches/mcQHrQutx

lisajamhoury, 2020-01-19:
Ok figured it out. It was actually an issue with VS Code. The editor was adding an import statement for text() at the top of my sketch file. 

import { text } from "body-parser";

It must have thought I wanted to import a module called text from body-parser rather than using the p5 text() functionality. My text editor is too smart for me. Haha.

closed 2020-01-19 by lisajamhoury