kinectron/kinectron #65: Multiframe no Image

thisiscwang, 2020-01-22:
There seem to be no image stream when using multiframe color and depth. When using the callbacks function, it just returns null with no image.

lisajamhoury, 2020-01-22:
Hi There, Please give some more details about the issue that you are having. 

—Which Kinect are you using? 
—Which Kinectron version / API version are you using?
—Are you working on one computer, or across a local network? 
—Are you working in the p5 web editor? Or with a local server? 
—Please include screenshots of any error messages in your developer console. 
—Please include any relevant code.
—Any other information you can include that might help me to understand the problem you are having.

Thanks!

thisiscwang, 2020-01-23:
Here are some more details with the current issue.
**Kinect Model**: Kinect V2
**Kinectron Version**: Kinectron Server 0.3.2 / Kinectron API 0.3.1
**Kinectron Setup**: One computer with a local server

- Code used to test initially, based off of [Multiframe ](https://editor.p5js.org/lisajamhoury/sketches/ryeHm7dGG)tutorial on the examples page:

```
var myCanvas = null;
var kinectron = null;

function setup() {
  myCanvas = createCanvas(1000, 1000);
  var kinectronIpAddress = "192.168.0.40";
  kinectron = new Kinectron(kinectronIpAddress);
  kinectron.makeConnection();
  kinectron.startMultiFrame(["color"], multiFrameCallback);
}

function draw() {
}

function multiFrameCallback(data) {
  loadImage(data.color, function(loadedImage) {
    image(loadedImage, 0, 0);
  });
}
```

- Error in the Console:

[image]

- Modified the `multiFrameCallback(data)` function to display the data into the console:

```
function multiFrameCallback(data)
{
  console.log(data);
  debugger;
}
```

- The Result from the Console:

[image]

lisajamhoury, 2020-01-28:
Thanks for the additional details. It is possible that Azure integration in version 0.3.0 broke the multiframe functionality. Could you try this earlier version of Kinectron which does not include Azure with the same code: 

Server: https://github.com/kinectron/kinectron/releases/tag/0.2.0
Client: https://cdn.jsdelivr.net/gh/kinectron/kinectron@0.2.0/client/dist/kinectron-client.js

In the meantime I will take a look at what could be causing the issue in the new server/client.

thisiscwang, 2020-02-06:
I have tested and it seems multiframe is working with version 0.2.0.

I have also tried modifying kinectron-app.js to get multiframe working and it seems to this change should work. In the function `drawImageToCanvas` starting at [line 1785](https://github.com/kinectron/kinectron/blob/6a98ef9aad8287081639371177b1859bba526b96/app/kinectron-app.js#L1785), I have added an additional condition at the if statement on [line 1790](https://github.com/kinectron/kinectron/blob/6a98ef9aad8287081639371177b1859bba526b96/app/kinectron-app.js#L1790). Below is the changes I have done:
```
function drawImageToCanvas(inCanvas, inContext, frameType, imageType, quality) {
  context.putImageData(imageData, 0, 0);
  inContext.clearRect(0, 0, inCanvas.width, inCanvas.height);
  inContext.drawImage(canvas, 0, 0, inCanvas.width, inCanvas.height);

  if (rawDepth || multiFrame) { // added multiFrame here
    return createDataUrl(inCanvas, frameType, imageType, quality);
  } else {
    createDataUrl(inCanvas, frameType, imageType, quality);
  }
}
```

One thing I have encountered with both Kinectron version (0.2 and 0.3) is that multiframe seems to lag or slow down overtime, after 10 seconds or so.

lisajamhoury, 2020-03-13:
Sorry for the delay. I just had a baby a few weeks ago and haven't had much time for my computer yet ;) But I will hopefully be able to take a look at this by next week. I will also look at where the lag is. There must be a memory leak somewhere. More soon...

playdscription, 2020-06-12:
I am having the same issue, will I have to revert to 0.20 for now or is there a fix?

lisajamhoury, 2020-06-13:
I should have the next release out with a fix in a week or two. But for now I'd recommend downgrading assuming you are working with Kinect 2. I'm working on implementing a fix but have not yet solved the memory leak.

lisajamhoury, 2020-07-10:
This bug is fixed in release 0.3.3 with a rewrite of the package and send functionality. 

I wasn't able to reproduce the multiframe slowing over time. I ran the multiframe  example with Windows Kinect for several minutes and it ran fine with no problem with memory. But I know the lag has happened to me in the past as I have issue #13 about it already. I'm closing this issue and will track progress on the lag on issue #13.

closed 2020-07-10 by lisajamhoury