kinectron/kinectron #77: Latency on Windows Kinect raw depth feed

lisajamhoury, 2020-07-10:
The Windows Kinect raw depth feed lags on the client side (in browser). The Kinectron server sends at around 25 fps, but the client receives at about 2 fps. I think the problem is that the amount of data from the server is overloading the client, because the problem is resolved when I send less data. For example, if I send only half the raw depth data, the sending and receiving fps is the same. Also, the Kinect Azure raw depth image, which is smaller, sends and receives without issue. 

@aaronsantiago put in place a system that keeps the server from sending additional data if there is already data in the data channel buffer (see [6ecbee3](https://github.com/kinectron/kinectron/commit/6ecbee3e14556e53874562c9ddd37e4fc348168fl)). This catches excessive data on the way out, but the client still gets overwhelmed. 

I did a number of tests on the client side and cannot find where the build up is happening. For now, I've reinstated the frame limiting in the Kinectron server to 15 fps on the raw depth feed as this keeps the client from getting overwhelmed (on my home network, at least).

It does look like there is a memory leak on the client that might be affecting the performance. It's possible the memory leak has to do with [setting img.src to dataUrl](https://stackoverflow.com/questions/19298393/setting-img-src-to-dataurl-leaks-memory). I will dig into this issue as a starting place because I cannot find where else the build up on the client is coming from.

wouterverweirder, 2021-02-13:
I ran into a similar memory leak when using data urls for image srcs - blob urls seem to work fine:

https://github.com/wouterverweirder/kinect-azure/blob/dee186167007d0fb53cf5f7004c103c3f6bb7a4e/examples/electron/renderer/demos/color-stream-jpeg.html#L59-L68

lisajamhoury, 2021-09-27:
I ([implemented blob urls](https://github.com/kinectron/kinectron/commit/7518e832c54e0e7a88d0b20f3d56032709f6bee9)) and am still running into the issue. I will integrate the blob urls anyway in an upcoming refactor bc of the known issues with dataurls. I will also try out a different tool for sending and receiving the data as it seems that the slow down is happening on the network as data is being received.

lisajamhoury, 2025-05-12:
I have released a new version of Kinectron Version 1.0.0 which resolves many of the issues. I'm closing this issue for now. If you still are running into the problem, please open a new issue.

closed 2025-05-12 by lisajamhoury