kinectron/kinectron #83: Review new azure examples

lisajamhoury, 2020-09-26:
@montoyamoraga I just added four new examples to the repo. I know you can't test with Azure Kinect, but if you have a moment would you please check over commenting, etc to make sure they are good to go? 

https://github.com/kinectron/kinectron/tree/master/examples/azure_examples/p5_examples/depthImage
https://github.com/kinectron/kinectron/tree/master/examples/azure_examples/p5_examples/depthKey
https://github.com/kinectron/kinectron/tree/master/examples/azure_examples/threejs_examples/pointCloud
https://github.com/kinectron/kinectron/tree/master/examples/azure_examples/threejs_examples/pointCloudDepthKey

thank you!

montoyamoraga, 2020-09-26:
hi @lisajamhoury 
i just went through the examples and i corrected some typos, they look great to me :)

closed 2020-09-26 by montoyamoraga

lisajamhoury, 2020-09-27:
@montoyamoraga excellent. thank you!