kinectron/kinectron #84: Skeleton Tracking Not Working on Azure Kinect

eraynai, 2020-10-06:
Using kinectron server-0.3.4 with the azure kinect. Unfortunately skeleton (tracked bodies), all bodies, or key is not working.  It is just blank when I click on skeleton and all bodies it is blank.  When I press key I get the color camera but the key feature is not working.

lisajamhoury, 2020-10-06:
Are you getting any errors in the console? 

Here are some things to try: 

Make sure you have the [Azure SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download) and [Body Tracking SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download) installed. 

Check that you have the [required hardware](https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements#body-tracking-host-pc-hardware-requirements) to run body tracking with Azure Kinect. The graphics card requirement is especially important and can cause the body tracking not to work. 

One simple thing: Try moving farther away from the Kinect and making a "[cactus pose](https://lh3.googleusercontent.com/proxy/kDF3kU1aSm6B99pFJan6w1I2tk9F8x8M61L11jro1VKLOxjYjTww2eEZSC-YZm_sbzk9ZFn-ZlD7nUw20NYaEqagNTvkClJoUzGbnPbKuWdb0m3XkykE1Qrflp7sw224qOg1Ledk6h9tHmTLZQ)" with your arms. Sometimes it takes the Kinect some movement to find the first body frame.

eraynai, 2020-10-06:
Yes, I was getting errors in the console.

WebSocket connection to 'ws://127.0.0.1:55670/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED

I installed the Azure SDK and confirmed that the Azure Kinect runs fine in their SDK tool.  I didn't install Body Tracking SDK, I will install that now and post how it goes.  Thanks!

lisajamhoury, 2020-10-06:
The WebSocket error is coming from peerjs, but shouldn't keep the body tracking (or any other feed) from running. 

The Body Tracking takes additional processing power, so let me know how that goes!

eraynai, 2020-10-06:
Ok, so I figured out the issue.  I have intel nuc with an amd graphics card that's equivalent to a 1060.  Requirements for body tracking are an NVIDIA GEFORCE GTX 1070 or better.  More specifically I need an Nvidia graphics card.  Oh well, I guess that solves that issue.  Thanks for your speed support, I will go back to kinect v2.

closed 2020-10-06 by eraynai

lisajamhoury, 2020-10-06:
Glad you figured it out, but sorry that it's not working. The GPU requirements for Azure are very limiting.