kinectron/kinectron #88: RBGD stream for Azure Kinect

vineet-ixd, 2021-02-26:


vineet-ixd, 2021-08-30:
Hello Lisa. Did you get a chance to work on this? Thanks

lisajamhoury, 2021-09-15:
Sorry I didn't. I started to look into it in March and I think there was an issue with one of the dependencies that was blocking me. I just put time on my calendar for this for early next week. I'll follow up in a week!

lisajamhoury, 2021-09-22:
@vineet-ixd I'm starting to work on this now and curious about your use case. Are you interested in the full RGBD image? Or are you looking for just the body index RGBD? I think I will try to implement both but the full RGBD is pretty bandwidth intensive, so I'm considering prioritizing the body index as I think it may be easier to send across a network.

lisajamhoury, 2021-09-27:
I've added the Azure RGBD in [release 0.3.7](https://github.com/kinectron/kinectron/releases/tag/0.3.7) feed with [an example](https://github.com/kinectron/kinectron/tree/master/examples/azure_examples/threejs_examples/rgbd) that uses [THREE-Kinectron](https://github.com/kinectron/Three-Kinectron).  

Make sure that you use both Kinectron application and client at 0.3.7. Let me know if you have any questions.

closed 2021-09-27 by lisajamhoury