kinectron/kinectron #92: adding IR feed to "multifeed" mode

garrett-laroy-johnson, 2021-04-26:
IR feeds can be very useful for doing CV with camera/projector set-ups. I see in Kinectron that Infrared is available for sending as a single stream. Is it possible to add this to the multi-stream mode as well?

lisajamhoury, 2021-05-01:
It seems that the underlying node module used in Kinectron, Kinect2 node, has not yet implemented Infrared in the multiSourceFrame. I've just submitted an issue there. If/when it is implemented, I will be able to make it available here as well. 

https://github.com/wouterverweirder/kinect2/blob/85af6c6aef3c4c8eda829d705af4fc50d3079ced/src/kinect2.cc#L1752
https://github.com/wouterverweirder/kinect2/issues/67

lisajamhoury, 2021-08-06:
Kinect2 node module now includes Infrared in the multiSourceFrame. I will look into integrating it into Kinectron before the end of August. 

https://github.com/wouterverweirder/kinect2/issues/67

lisajamhoury, 2021-09-21:
Circling back on this. Adding Infrared requires updating to kinect2 v2 (^0.2.1). I'm running into a weird bug with the upgrade on the RGB feed (see #100). Once I am able to resolve that bug I can complete this.

lisajamhoury, 2025-05-12:
Thank you for posting this issue. I have released a new version of Kinectron Version 1.0.0 which resolves many of the issues. I'm closing this issue for now. If you still are running into the problem, please open a new issue.

closed 2025-05-12 by lisajamhoury