kinectron/kinectron #103: Uncaught Error: The specified module could not be found.

MandyLiuu, 2021-10-09:
kinectron version: 0.3.7
OS: windows 10
SDK: kinect for windows SDK v2.0

when I try to run kinect windows v2, meet this issue:
[image]

lisajamhoury, 2021-10-09:
Could you try deleting 0.3.7 and re-downloading 0.3.7? Sometimes this happens because of an issue with the download.

lisajamhoury, 2021-10-09:
Nevermind I was able to reproduce the error on my end. Will look into it now.

lisajamhoury, 2021-10-09:
Ok I fixed this by rebuilding kinect2 v0.1.7 for electron as the build was missing in node_modules. I've updated the download link here https://github.com/kinectron/kinectron/releases/tag/0.3.7

Here's [a direct link to the updated app](https://www.dropbox.com/s/sdn095j349kru48/Kinect-server-0.3.7.zip?dl=0).

@MandyLiuu Lmk if this fix works for you.

MandyLiuu, 2021-10-11:
Great, version 0.3.7 can now be used in my environment. 
@lisajamhoury Thank u for ur patience and efforts.

flololan, 2022-05-16:
Hi @lisajamhoury ,

I have the same error using your Dropbox link. What should I do?

[image]

flololan, 2022-05-17:
Reinstalling the SDK for Kinect V2 fixed it.

commit a0588471 (2021-10-09): fixes #103