kinectron/kinectron #108: colored_point_cloud

badcode06, 2022-04-04:
hi lisa, thanks for this great works.
I have a small question. I'm trying to find a demo for colored point cloud (https://vimeo.com/212533472?embedded=true&source=video_title&owner=12874561) but still couldn't find. i tried every example. do you have any documentation or link for it?
and can you explain more about how to run examples? 
I mean I'm running kinectron_server, and also creating a local server for example? so is it okay? should I do something?

lisajamhoury, 2022-04-04:
@badcode06 this is an excellent question. Actually I made that example five years ago for my graduate thesis research. 

I think [this is the code](https://github.com/kinectron/kinectron/tree/documentation/examples/threejs_examples/finalOneShader). 

It is very messy and not commented at all. Also, I think it will not be possible to run that example with the current release of Kinectron. I was running experimental versions at the time to make examples for my thesis project. 

You can also look through the examples [here](https://github.com/kinectron/kinectron/tree/documentation/examples/threejs_examples) from around that same time of experimentation. They also will not work with the current release, but you can see the different techniques I was experimenting with at the time which might give you some ideas if you want to experiment on your own :) 

I've gone through all of the examples that I have and it's true I haven't released an example that adds color to the point cloud. One of the reasons is because it is very network intensive to send both the color data and the depth data at the same time. However, it should be possible to create an example for you. It may not be very performant, but I can show you how I would do it. Let me see if I can find time to take a look at this at the end of this week. I'll update my progress here.

badcode06, 2022-04-11:
huh thanks, network bandwidth is no problem for now, just want to add color information to clouds, I'm looking forward to the updates

badcode06, 2022-04-22:
@lisajamhoury hi Lisa, are there any updates?

badcode06, 2022-05-09:
hii, @lisajamhoury actually, the webrtc side is not important for me now. I mean the feed can be just two images, depth, and RGB and I'm trying to understand how to create colored point clouds on browsers with these two images. after that I can concentrate on the streaming side but as I said it's not vital now.  when I try to use depth images results are not successful yet, callbacks make me confused a little bit. so im waiting have a nice week. cheers.

lisajamhoury, 2023-03-20:
hi @badcode06 sorry I haven't looked at it yet. I'm just back from parental leave. I put a block next week on my calendar. I will try to run this then and will update the thread. thanks for the reminder.

lisajamhoury, 2025-05-12:
Thank you for posting this issue. I have released a new version of Kinectron Version 1.0.0 which resolves many of the issues. I'm closing this issue for now. If you still are running into the problem, please open a new issue.

closed 2025-05-12 by lisajamhoury